DSheirer/sdrtrunk #2165: Tuner doesn't start on Linux unless I run as root

PhillyPhoto, 2025-03-01:
**sdrtrunk Version**
0.6.1 'master-branch'

**Describe the bug**
I'm given an error saying "access denied" for the tuner. I've attempted the udev rules steps in the wiki, but they don't make a difference. The only way to get the tuners to start is to launch the app as root.

**To Reproduce**
Steps to reproduce the behavior:
1. Open app in Linux
2. Click on Tuners
4. See error

**Expected behavior**
The tuners should start when the app is launched automatically, or when the rules are configured at minimum.

**Screenshots**
Without sudo:
[image]

With sudo:
[image]

**Application Log**
No sudo:
[sdrtrunk_app_no_sudo.log](https://github.com/user-attachments/files/19037025/sdrtrunk_app_no_sudo.log)

Sudo:
[sdrtrunk_app_sudo.log](https://github.com/user-attachments/files/19037027/sdrtrunk_app_sudo.log)

**Desktop (optional - complete the following information):**
 - OS: Linux Mint
 - CPU Cores: 4
 - RAM: 16GB

**Additional context**
I've tried using chown to own the rules.d directory and sub files to no avail as well.

Randesign, 2025-03-02:
Did you install `sudo apt install rtl-sdr`

PhillyPhoto, 2025-03-02:
> Did you `sudo apt install rtl-sdr` ?

That may have done it. It's streaming now. Thanks!

closed 2025-03-02 by PhillyPhoto