DSheirer/sdrtrunk #2211: RTL-SDR

the-geth, 2025-03-24:
[20250317_sdrtrunk_app.log](https://github.com/user-attachments/files/19413049/20250317_sdrtrunk_app.log)

**sdrtrunk Version**
sdrtrunk v0.6.1

**Describe the bug**
RTL-2832/R820T- tuner is visible in tuner selection however, when in Polyphase, Playlist editor says "No Tuner Available"
when in heterodyne, channel plays but does not capture signals, only displays messages of "SYNC LOSS - BITS PROCESSED [1570]

**Desktop (optional - complete the following information):**
 - OS: Kubuntu 24.04
 - CPU Cores: 6
 - RAM:16gb 

**Additional context**
Add any other context about the problem here.

[image]

[image]

DSheirer, 2025-03-24:
Are there any errors in the application log file?  

In your first screenshot where you're using the polyphase.  If you select the tuner in the Tuners tab and click the 'View Spectrum' button, does the tuner show samples in the spectral display and waterfall?

the-geth, 2025-03-24:
> Are there any errors in the application log file?
>  I provided the log file generated yesterday in this post, heres alot of the errors im seeing:

20250323 173216.665 [main] ERROR i.g.d.a.playback.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [NVidia [plughw:0,3]]  [10MB/28MB 37%]
20250323 173444.381 [sdrtrunk cached thread 3] ERROR i.g.d.s.t.m.PolyphaseChannelSourceManager - Error getting source  [105MB/158MB 66%]
org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
20250323 183519.600 [sdrtrunk cached thread 4] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Couldn't allocate channel. Requested channel cannot be provided by this channelizer.  Requested channel [856206250 - 856218750] exceeds current channelizer frequency range [99900000 - 102300000] [Log Suppress 1/3]  [112MB/136MB 82%]
20250323 183519.600 [sdrtrunk cached thread 4] ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [112MB/136MB 82%]
20250323 183801.127 [sdrtrunk cached thread 5] ERROR i.g.d.d.f.c.PolyphaseChannelManager - Couldn't allocate channel. Requested channel cannot be provided by this channelizer.  Requested channel [856206250 - 856218750] exceeds current channelizer frequency range [99900000 - 102300000] [Log Suppress 2/3]  [121MB/148MB 82%]
20250323 183801.128 [sdrtrunk cached thread 5] ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [121MB/148MB 82%]
20250323 183845.243 [sdrtrunk cached thread 3] ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Control] - No Tuner Available  [93MB/164MB 57%]

> In your first screenshot where you're using the polyphase. If you select the tuner in the Tuners tab and click the 'View Spectrum' button, does the tuner show samples in the spectral display and waterfall?

yes i can see spectrum waterfall, as you can see in this screenshot

[20250323_sdrtrunk_app.log](https://github.com/user-attachments/files/19431215/20250323_sdrtrunk_app.log)

[image]

Jedi425, 2025-03-25:
I am seeing similar behavior on my system. I also see waterfall when I enable the tuner, but with nothing running it will not consistently start the channel.

**sdrtrunk Version**
sdrtrunk v0.6.1 and -beta-1 (tried both)

**Log File**
[sdrtrunk_app.log](https://github.com/user-attachments/files/19453521/sdrtrunk_app.log)

**Describe the bug**
See above.

**Desktop (optional - complete the following information):**

    OS: Arch Linux
    CPU Cores: 6
    RAM:64gb

I have attached a log file of me starting the application, attempting to start a channel, and then shutting down.

Randesign, 2025-03-26:
It could be that the USB device is read only. Did you:
`sudo apt install rtl-sdr`

Jedi425, 2025-03-26:
I have done so, no change. That was the correct command, though, so that worked.

Other radio apps are working fine; I used gqrx just now, in fact. And sdrtrunk can see the SDR; it gets a waterfall up. It just won't start a channel.

Randesign, 2025-03-27:
In both of these cases, there is an audio error that appears first in the logs: 
`Couldn't obtain audio source data line for audio output - mixer`
I don't know if that's something which later causes a USB pipe error. I guess I would start with getting the audio error figured out. 
Have either of you tried selecting a different audio device?
It's as if something on your system is interfering with java being able to write to your hardware devices. I don't think this is a specific bug with sdrtrunk. Perhaps try other java based apps that access audio and USB ports.

Jedi425, 2025-03-27:
The 'test' buttons all work in User Preferences; all beeps sound. Wouldn't that also fail if the app couldn't write to the audio hardware? And at any rate, those lines are for an audio source with nothing plugged into it:

20250326 210852.621 [JavaFX Application Thread] ERROR i.g.d.a.playback.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [HDMI [plughw:0,3]]  [78MB/168MB 46%]
20250326 210855.007 [JavaFX Application Thread] ERROR i.g.d.a.playback.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [HDMI [plughw:0,7]]  [120MB/168MB 71%]
20250326 210857.306 [JavaFX Application Thread] ERROR i.g.d.a.playback.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [HDMI [plughw:0,8]]  [98MB/168MB 58%]
20250326 210859.650 [JavaFX Application Thread] ERROR i.g.d.a.playback.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [HDMI [plughw:0,9]]  [78MB/168MB 46%]

I'm not using the HDMI connection for audio. I'm using my onboard sound through ALSA.

[sdrtrunk_app.log](https://github.com/user-attachments/files/19479353/sdrtrunk_app.log)

I've attached today's log. In it is this excerpt, which is a full cycle of me opening the app, the app failing, and me closing the app:

`20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.1  [9MB/32MB 30%]
20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.10  [9MB/32MB 30%]
20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2024-12-03T16:31:53.486-0500  [9MB/32MB 30%]
20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog - Build-JDK         : 23 (BellSoft 23+38  [9MB/32MB 30%]
20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.8.0-49-generic  [9MB/32MB 30%]
20250326 211146.962 [main] INFO  i.g.d.log.ApplicationLog -   [9MB/32MB 30%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/32MB 30%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [9MB/32MB 30%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [10MB/32MB 31%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [10MB/32MB 31%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [10MB/32MB 31%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [10MB/32MB 31%]
20250326 211146.963 [main] INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog - Host OS Version:       6.13.8-arch1-1  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        16  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  15 GB  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog - Storage Directories:  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog -  Application Root: /home/Jedi425/SDRTrunk  [10MB/32MB 31%]
20250326 211146.964 [main] INFO  i.g.d.log.ApplicationLog -  Application Log:  /home/Jedi425/SDRTrunk/logs  [10MB/32MB 31%]
20250326 211146.965 [main] INFO  i.g.d.log.ApplicationLog -  Event Log:        /home/Jedi425/SDRTrunk/event_logs  [10MB/32MB 31%]
20250326 211146.965 [main] INFO  i.g.d.log.ApplicationLog -  Playlist:         /home/Jedi425/SDRTrunk/playlist  [10MB/32MB 31%]
20250326 211146.965 [main] INFO  i.g.d.log.ApplicationLog -  Recordings:       /home/Jedi425/SDRTrunk/recordings  [10MB/32MB 31%]
20250326 211146.976 [main] INFO  i.g.d.s.t.s.a.SDRPlayLibraryHelper - SDRPLay API library loaded by name [sdrplay_api]  [10MB/32MB 33%]
20250326 211146.976 [main] INFO  i.github.dsheirer.gui.SDRTrunk - SDRPlay API native library preemptively loaded  [10MB/32MB 33%]
20250326 211147.029 [main] INFO  i.g.dsheirer.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [13MB/32MB 42%]
20250326 211147.030 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/home/Jedi425/SDRTrunk/SDRTrunk.properties]  [13MB/32MB 42%]
20250326 211147.182 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [8MB/48MB 17%]
20250326 211147.187 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.262  [8MB/48MB 17%]
20250326 211147.187 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.22.11312  [8MB/48MB 17%]
20250326 211147.192 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [22] potential usb devices  [8MB/48MB 17%]
20250326 211147.195 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [1] Port [8.1] Tuner Class [RTL-2832]  [9MB/48MB 19%]
20250326 211147.197 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:1 Port:8.1 - Added / Starting ...  [9MB/48MB 19%]
20250326 211149.145 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [13MB/48MB 29%]
[... cut ...]
20250326 211727.173 [AWT-EventQueue-0] INFO  i.github.dsheirer.gui.SDRTrunk - Stopping tuners ...  [352MB/536MB 65%]
20250326 211727.173 [AWT-EventQueue-0] INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:1 Port:8.1  [352MB/536MB 65%]
20250326 211727.177 [AWT-EventQueue-0] INFO  i.github.dsheirer.gui.SDRTrunk - Shutdown complete.  [354MB/536MB 66%]
20250326 211727.177 [AWT-EventQueue-0] INFO  i.g.d.log.ApplicationLog - Stopping application logging  [354MB/536MB 66%]`

No mixer errors there. I think the mixer issue is unrelated. Again, the 'test' buttons all produce beeps when set to ALSA.

DSheirer, 2025-03-27:
> [20250317_sdrtrunk_app.log](https://github.com/user-attachments/files/19413049/20250317_sdrtrunk_app.log)
>
> **sdrtrunk Version** sdrtrunk v0.6.1
>
> **Describe the bug** RTL-2832/R820T- tuner is visible [...]

@the-geth In your original post in the applciation log is this entry:

20250317 191755.370 [AWT-EventQueue-0] ERROR i.g.d.s.t.u.USBTunerController - Access to USB tuner denied - (windows) reinstall zadig driver or (linux) blacklist driver and/or check udev rules  [39MB/136MB 28%]

Make sure you follow the setup instructions in the users manual.

blainek-ew, 2025-04-07:
For what it's worth, I had to re-download the JMBE audio library and let it rebuild after SDRtrunk was installed.  For some reason the initial build did not work even though JMBE was the same version.

Second thing I noted but maybe not related is that I had to connect my USB Hub into another powered hub to get the SDR radios to connect properly.  I am still trying to determine why this is the case as both hubs are powered, but play around a bit with hubs and ports to see if the tuner will wake up.  also make sure you have the latest zadiag and reinstall the USB driver selecting the proper device in the selection list for your tuner.

the-geth, 2025-04-24:
I am now trying to use SDRtrunk in a windows 10 virtual machine.
Same problem. the RTL-SDR works fine with SDR sharp. i have re-installed zadig and JMBE multiple times, still same "SYNC LOSS - BITS PROCESSED" message.
is there any info you would like me to provide?

blainek-ew, 2025-04-24:
This usually happens when the signal is not strong enough or the offset is
too far off center.   Make sure you have the control frequency centered
properly.

profporridge, 2025-05-20:
> I am now trying to use SDRtrunk in a windows 10 virtual machine. 

Are you running the nightly build? As mentioned above there is a fix for RTL-SDR on windows #1963 that is scheduled for 0.6.2