DSheirer/sdrtrunk #2213: Unsupported SDRPlay API version on 0.6.1-Beta 3 and 0.6.1 (final)

davebelle85, 2025-04-05:
**sdrtrunk Version**
0.6.1-Beta 3 and 0.6.1 (final)

**Describe the bug**
I have installed SDRPlay API version 3.15 but in the console I get the error "API library is not available - unsupported version 0.0"

**Screenshots**

[image]

[image]

**Desktop (optional - complete the following information):**
 - OS: Windows 11
 - CPU Cores: I7 20 cores
 - RAM: 64 GB

Thanks

DSheirer, 2025-04-05:
Please attach your sdrtrunk application log so that I can see the whole startup sequence.

Also, can you please try restarting the SDRPlay API server via the service manager and then launch sdrtrunk and see if you get the same results?

DSheirer, 2025-04-05:
If you're comfortable modifying the code and trying out a patch, can you update the following code and see if that resolves the issue?

Class: SDRPlayLibraryHelper Line: 44

```
    private static final String SDRPLAY_API_PATH_WINDOWS = System.getenv("ProgramFiles") +
            "\\SDRplay\\API\\x64\\" + SDRPLAY_API_LIBRARY_NAME + ".dll";
```

This just adds the windows library extension (.dll) to the end of the expected path so that it can find the file and load it.

davebelle85, 2025-04-05:
Yep that did it. Thanks :)

closed 2025-04-05 by davebelle85