DSheirer/sdrtrunk #2228: RSP-DUO turns off

ah1102, 2025-05-09:
At 17:23, I started listening to two channels. But at 17:33, the listening was interrupted and the receiver seemed to disconnect. This has been observed for quite a while now.

OS: _macOS 11 BigSur_
Programm: _sdr-trunk-osx-x86_64-vnightly_ and _sdr-trunk-osx-x86_64-v0.6.1_
SDR: SDRplay RSP-duo

```
2025-05-09 17:23:30.541 INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [2] channels now - user invoked  [109MB/148MB 74%]
2025-05-09 17:23:30.631 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [/Users/Apple/SDRTrunk/jmbe/jmbe-1.0.9.jar]  [116MB/148MB 78%]
2025-05-09 17:23:30.647 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [121MB/148MB 81%]
2025-05-09 17:23:30.647 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [121MB/148MB 81%]
2025-05-09 17:23:30.684 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [5000000.0] providing [200] channels at [25000.0] Hz each  [122MB/148MB 82%]

2025-05-09 17:33:35.615 INFO  i.g.d.s.t.s.RspTunerController - Processing device removal for tuner [TUNER 1] controller class [RspDuoTuner1Controller]  [589MB/1000MB 58%]
2025-05-09 17:33:35.615 INFO  i.g.d.s.t.u.DiscoveredTunerModel - Tuner removal detected - stopping and removing: RSPduo Tuner 1 SER#******  [590MB/1000MB 59%]
2025-05-09 17:33:35.616 INFO  i.g.d.s.t.u.DiscoveredTunerModel - Removing discovered tuner: RSPduo Tuner 1 SER#*****  [590MB/1000MB 59%]
2025-05-09 17:33:35.616 INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RSPduo Tuner 1 SER#*****  [590MB/1000MB 59%]
2025-05-09 17:33:35.618 INFO  i.g.d.m.d.p.P25TrafficChannelManager - Stopping traffic channel: P25-1_Site 001-009_0  [590MB/1000MB 59%]
2025-05-09 17:33:35.619 INFO  i.g.d.m.d.p.P25TrafficChannelManager - Stopping traffic channel: P25-1_Site 001-017_1  [590MB/1000MB 59%]
```

ah1102, 2025-05-09:
When receiving one channel AM, a processor is loaded by 120%. All four cores of the CPU

[image]

[image]

DSheirer, 2026-11-07:
Closing.  sdrtrunk now uses new jSDRPlay driver library.

closed 2026-11-07 by DSheirer