DSheirer/sdrtrunk #2230: Playlist editor sorting function

ah1102, 2025-05-10:
Added playlists are sorted by date of addition. In the night version of the program, the sorting by "favorites" and "playlist" doesn't seem to work. I would like to have the option for more flexible sorting, something like a custom sorting feature where playlists could be manually rearranged. Because right now, there are only two sorting options available.

[image]

DSheirer, 2026-11-08:
I'm able to click the 'Playlist' column header and apply: a)sort forward, b) sort revers, c) remove the column sort.

It seems to be working as intended.  Closing

closed 2026-11-08 by DSheirer

fredcogit, 2026-11-08:
Hello Denny,

Yes, I noted that 2230 was resolved when ny issue 2342 titled "Playlists in the Playlist column on the Playlists tab in the Playlist Editor not sorting when clicking on column header" was completed and closed on July 12, 2026.

I have not tried any nightlys after that, but today I installed the new 1.0.0 Beta 1, and I see that the ability to sort the playlists IS working in this version...however I see that the sort order I have chosen (alphabetical, asending...upward-facing arrow) is NOT being retained across program shutdown and restart.  

Would you consider modifying the "fix from 2342" so that the sort order chosen IS retained and persists across shutdowns/restarts? 

Thank you so much!

Regards,

Fred