DSheirer/sdrtrunk #2232: Frequency 158.775 error no tuner available

StrawberryMew, 2025-05-12:
**sdrtrunk Version**
master-branch

**Describe the bug**
Frequency 158.775 error no tuner available

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'playlist editor'
2. Click on 'play on that frequency'
3. See error

**Expected behavior**
To be able to listen on that frequency.

**Screenshots**

[image]

**Application Log**
2025-05-11 21:31:22.439 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [158775000] from preferred tuner [RTL-2832/R828D 00000001] - searching for another tuner  [222MB/256MB 86%]
2025-05-11 21:31:22.440 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [] - No Tuner Available  [222MB/256MB 86%]

**Additional context**
The error will show even if i am not listening to any other stations.

TravisSamren, 2025-05-12:
What bug? How many sdr's do you have? Buy more?

StrawberryMew, 2025-05-12:
It doesn't matter if I have no frequencies running or multiple frequencies running it so gives the same error

TravisSamren, 2025-05-12:
Oh hmm. From the pic i see two playing, but also 3 idle in background looks like. But from the pic you provide it shows 2 channels playing when you trying to start a third. Also shows you have a preferred tuner set. Just wondering how many sdr's you have hooked up?

StrawberryMew, 2025-05-12:
Just 1 sdr-rtl v4

DSheirer, 2025-05-12:
Can you please include the whole application log that includes all of your tuners and shows the error that you're seeing?

Check that you haven't adjusted the minimum and maximum frequency values for your tuners where the min-max range might be preventing that frequency from being used.

StrawberryMew, 2025-05-12:
WARNING: Using incubator modules: jdk.incubator.vector
2025-05-12 09:10:20.038 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\Hana\SDRTrunk\settings\icons.xml]  [16MB/500MB 3%]
2025-05-12 09:10:20.042 INFO  i.g.d.icon.IconModel - Icons file not found at [C:\Users\Hana\SDRTrunk\settings\icons.xml]  [16MB/500MB 3%]
2025-05-12 09:10:20.312 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\Hana\SDRTrunk\logs\sdrtrunk_app.log  [6MB/500MB 1%]
2025-05-12 09:10:20.337 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : nightly  [9MB/500MB 1%]
2025-05-12 09:10:20.337 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.10  [9MB/500MB 1%]
2025-05-12 09:10:20.337 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2025-04-20T12:29:56.615+0000  [9MB/500MB 1%]
2025-05-12 09:10:20.337 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 23.0.2 (BellSoft 23.0.2+9  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.8.0-1021-azure  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog -   [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/500MB 1%]
2025-05-12 09:10:20.338 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 11  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        16  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [9MB/500MB 1%]
2025-05-12 09:10:20.339 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [9MB/500MB 1%]
2025-05-12 09:10:20.340 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\Hana\SDRTrunk  [9MB/500MB 1%]
2025-05-12 09:10:20.340 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\Hana\SDRTrunk\logs  [9MB/500MB 1%]
2025-05-12 09:10:20.340 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\Hana\SDRTrunk\event_logs  [9MB/500MB 1%]
2025-05-12 09:10:20.340 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\Hana\SDRTrunk\playlist  [9MB/500MB 1%]
2025-05-12 09:10:20.341 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\Hana\SDRTrunk\recordings  [9MB/500MB 1%]
2025-05-12 09:10:20.355 INFO  i.g.d.s.t.s.a.SDRPlayLibraryHelper - SDRPlay API native library not found at: C:\Program Files\SDRplay\API\x64\sdrplay_api  [6MB/40MB 16%]
2025-05-12 09:10:20.364 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [7MB/40MB 18%]
2025-05-12 09:10:20.370 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\Hana\SDRTrunk\SDRTrunk.properties]  [7MB/40MB 18%]
2025-05-12 09:10:20.509 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [23MB/40MB 59%]
2025-05-12 09:10:20.576 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [25MB/40MB 62%]
2025-05-12 09:10:20.577 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [25MB/40MB 62%]
2025-05-12 09:10:20.599 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [32] potential usb devices  [25MB/40MB 62%]
2025-05-12 09:10:20.602 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [5] Port [8.4] Tuner Class [RTL-2832]  [25MB/40MB 63%]
2025-05-12 09:10:20.604 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:5 Port:8.4 - Added / Starting ...  [25MB/40MB 63%]
2025-05-12 09:10:21.691 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [10MB/40MB 27%]
2025-05-12 09:10:21.879 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [11MB/40MB 27%]
2025-05-12 09:10:21.881 INFO  i.g.d.s.t.s.api.SDRplay - API library is not available - unsupported version: 0.0  [11MB/40MB 27%]
2025-05-12 09:10:21.883 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\Hana\SDRTrunk\settings\settings.xml]  [11MB/40MB 28%]
2025-05-12 09:10:21.931 INFO  i.g.d.m.DiagnosticMonitor - Diagnostic monitoring enabled running every 30 seconds  [16MB/40MB 40%]
2025-05-12 09:10:22.488 INFO  i.g.d.p.PlaylistManager - Loading playlist [C:\Users\Hana\SDRTrunk\playlist\default.xml]  [27MB/40MB 67%]
2025-05-12 09:10:22.537 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [17MB/40MB 44%]
2025-05-12 09:10:28.056 INFO  i.g.d.s.r.RadioReference - Radio Reference Account - User:StrawberryMew Expires:05-06-2026 Login Status:VALID_PREMIUM Premium Access Enabled:true  [65MB/88MB 74%]
2025-05-12 09:10:33.076 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [C:\Users\Hana\SDRTrunk\jmbe\jmbe-1.0.9.jar]  [175MB/296MB 59%]
[... cut ...]
2025-05-12 09:10:33.083 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [176MB/296MB 59%]
2025-05-12 09:10:33.095 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [177MB/296MB 59%]

this is with it by itself aparently it working like that today 

2025-05-12 09:12:13.315 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Pampa FD Dispatc] - No Tuner Available  [151MB/296MB 51%]

StrawberryMew, 2025-05-12:
this is the other way around

WARNING: Using incubator modules: jdk.incubator.vector
2025-05-12 09:14:26.198 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\Hana\SDRTrunk\settings\icons.xml]  [16MB/500MB 3%]
2025-05-12 09:14:26.201 INFO  i.g.d.icon.IconModel - Icons file not found at [C:\Users\Hana\SDRTrunk\settings\icons.xml]  [17MB/500MB 3%]
2025-05-12 09:14:26.461 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\Hana\SDRTrunk\logs\sdrtrunk_app.log  [6MB/500MB 1%]
2025-05-12 09:14:26.486 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : nightly  [9MB/500MB 1%]
2025-05-12 09:14:26.486 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.10  [9MB/500MB 1%]
2025-05-12 09:14:26.486 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2025-04-20T12:29:56.615+0000  [9MB/500MB 1%]
2025-05-12 09:14:26.486 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 23.0.2 (BellSoft 23.0.2+9  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.8.0-1021-azure  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog -   [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/500MB 1%]
2025-05-12 09:14:26.487 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [9MB/500MB 1%]
2025-05-12 09:14:26.488 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 11  [9MB/500MB 1%]
2025-05-12 09:14:26.488 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [9MB/500MB 1%]
2025-05-12 09:14:26.488 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [9MB/500MB 1%]
2025-05-12 09:14:26.488 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        16  [9MB/500MB 1%]
2025-05-12 09:14:26.488 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [9MB/500MB 1%]
2025-05-12 09:14:26.489 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [9MB/500MB 1%]
2025-05-12 09:14:26.489 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\Hana\SDRTrunk  [9MB/500MB 1%]
2025-05-12 09:14:26.489 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\Hana\SDRTrunk\logs  [9MB/500MB 1%]
2025-05-12 09:14:26.489 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\Hana\SDRTrunk\event_logs  [9MB/500MB 1%]
2025-05-12 09:14:26.490 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\Hana\SDRTrunk\playlist  [9MB/500MB 1%]
2025-05-12 09:14:26.490 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\Hana\SDRTrunk\recordings  [9MB/500MB 1%]
2025-05-12 09:14:26.513 INFO  i.g.d.s.t.s.a.SDRPlayLibraryHelper - SDRPlay API native library not found at: C:\Program Files\SDRplay\API\x64\sdrplay_api  [6MB/40MB 16%]
2025-05-12 09:14:26.522 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [7MB/40MB 17%]
2025-05-12 09:14:26.528 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\Hana\SDRTrunk\SDRTrunk.properties]  [7MB/40MB 17%]
2025-05-12 09:14:26.667 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [23MB/40MB 59%]
2025-05-12 09:14:26.736 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [25MB/40MB 62%]
2025-05-12 09:14:26.736 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [25MB/40MB 62%]
2025-05-12 09:14:26.754 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [32] potential usb devices  [25MB/40MB 62%]
2025-05-12 09:14:26.757 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [5] Port [8.4] Tuner Class [RTL-2832]  [25MB/40MB 63%]
2025-05-12 09:14:26.759 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:5 Port:8.4 - Added / Starting ...  [25MB/40MB 63%]
2025-05-12 09:14:27.838 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [10MB/40MB 26%]
2025-05-12 09:14:28.015 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [10MB/40MB 27%]
2025-05-12 09:14:28.016 INFO  i.g.d.s.t.s.api.SDRplay - API library is not available - unsupported version: 0.0  [10MB/40MB 27%]
2025-05-12 09:14:28.019 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\Hana\SDRTrunk\settings\settings.xml]  [10MB/40MB 27%]
2025-05-12 09:14:28.071 INFO  i.g.d.m.DiagnosticMonitor - Diagnostic monitoring enabled running every 30 seconds  [16MB/40MB 40%]
2025-05-12 09:14:28.708 INFO  i.g.d.p.PlaylistManager - Loading playlist [C:\Users\Hana\SDRTrunk\playlist\default.xml]  [26MB/40MB 67%]
2025-05-12 09:14:28.758 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [17MB/40MB 43%]
2025-05-12 09:14:39.404 INFO  i.g.d.s.r.RadioReference - Radio Reference Account - User:StrawberryMew Expires:05-06-2026 Login Status:VALID_PREMIUM Premium Access Enabled:true  [61MB/96MB 63%]
[... cut ...]
2025-05-12 09:14:43.867 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [154MB/216MB 71%]

then when i add fire to it

2025-05-12 09:15:22.896 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [158775000] from preferred tuner [RTL-2832/R828D 00000001] - searching for another tuner  [105MB/216MB 48%]
2025-05-12 09:15:22.897 ERROR i.g.d.g.p.c.ChannelConfigurationEditor - Error starting channel [Gray Fire North] - No Tuner Available  [105MB/216MB 48%]

DSheirer, 2025-05-12:
If you only have a single tuner and that tuner is already providing a channel frequency, then the tuner can't source another frequency that is outside of the 2.4 MHz bandwidth.  In your screenshots, you have channels running at 154 MHz at the same time that you're trying to start a channel at 158 MHz.  The spread between those two channels (158 - 154 = 4MHz) exceeds the spread of a 2.4 MHz tuner bandwidth. 

If you have multiple tuners, make sure that each tuner's minimum and maximum frequency values have not been changed, or click 'reset' to restore the min/max frequency values so that the tuner can source the frequency you're trying to listen to.  If you've specified a min-max range that doesn't allow for your 158 MHz channel, then the tuner isn't going to be able to provide it.

When you see this error message: 
```
2025-05-12 09:18:34.390 INFO i.g.d.s.t.m.TunerManager - Unable to source channel [156150000] from preferred tuner [RTL-2832/R828D 00000001] - searching for another tuner [136MB/200MB 68%]
```
This can sometimes happen when you have selected a 'preferred' tuner for a channel and that tuner is no longer available.  As the error message indicates, the software attempts to find an another tuner to source the channel.  You can resolve the logging of this message by clearing the preferred tuner from the channel and starting it normally.  However, this doesn't solve the problem if there is still no tuner available that can source the channel.

DSheirer, 2025-05-12:
sdrtrunk only uses 90% of the tuner sample rate/bandwidth, so for a setting of 3.2 MHz, only 90% (2.88 MHz) of that bandwidth is usable.

DSheirer, 2025-05-14:
Is there still an issue or do you intend to close this?

closed 2025-05-14 by StrawberryMew