DSheirer/sdrtrunk #2236: Poor Signal Resolution When Using SDR Trunk As Compared to SDR# w/RTL-SDR v3

BigSauce490, 2025-05-31:
sdr-trunk-windows-x86_64-v0.6.1

**Poor Signal Resolution When Using SDR Trunk As Compared to SDR# w/RTL-SDR v3**

Recreate: Tune to station with known quality and observe clean signal in SDR#. Repeat with SDR Trunk with identical hardware and antenna setup, then observe drastically reduced signal resolution/clarity.

I recently purchased two RTL-SDR v3s and wanted to listen in on my local P25 Phase 1 sites. I spent some time getting things set up and was able to track and achieve sync on the P25 control signal with no issues, but when I plugged in a second dongle I noticed plenty of "events" being recorded on the primary window but no audio. Further troubleshooting using the spectrum analyzer I noticed that the dongle being used to hop between channels had very poor resolution compared to the dongle being used to monitor the control channel. This appeared to cause the scanner to randomly hop around and essentially listen to noise thinking it was data.

At first I figured that dongle #2 was defective, but to double check I hopped into SDR# to compare the two and saw no real difference in signal clarity, but when I went into SDR trunk and tuned to the same frequency I noticed drastically different results.

Attached are two screenshots, both using the same dongle, same cable, same USB port, and same antenna - the only difference is the software being used to view the spectrum. You'll notice that SDR# looks fine, but SDR Trunk looks downright bad for some reason....

 - OS: Windows 10
 - CPU Cores: 6 cores
 - RAM: 32 GB

[image]

[image]

BigSauce490, 2025-05-31:
Note: In the SDR Trunk screenshot, the scanner was running through the different primary and alternate control channels so that's why the "Frequency" field doesn't match the SDR# frequency. Just know that the control signal of interest is 858.1375 MHz as seen in the SDR # screenshot. Just looking at the two signal bandwidths you'll see that SDR# has it at about 12.5 kHz while SDR Trunk sees something like 600 kHz.

DSheirer, 2025-05-31:
Don't use automatic gain on these tuners.

You need to adjust the PPM and adjust the gain in EACH OF YOUR TUNERs.  You may be able to make that inteferring FM radio signal disappear with the gain adjustments, but if not, recommend you get an inline FM notch filter.

In your screenshot, it looks like the PPM is way off with the signals appearing to the left (ie lower frequency) relative to your channel overlays.

closed 2025-05-31 by BigSauce490