DSheirer/sdrtrunk #2238: Using 2 Airspys, not locking onto control channels

N6MIK, 2025-06-17:
**sdrtrunk Version**
0.6.1 Nightly and Final 

**Describe the bug**
When using 2 Airspy dongles (1 mini, 1 R2), SDRTrunk scans control channels frequencies but does not land on any or only receives for a few seconds. 

Error in logfile is: 
2025-06-17 07:34:51.949 ERROR i.g.d.sample.Broadcaster - Error while broadcasting [class io.github.dsheirer.dsp.symbol.Dibit] to listeners [Log Suppress 1/5]  [218MB/390MB 56%]
java.lang.NullPointerException: Cannot invoke "io.github.dsheirer.module.decode.p25.phase1.P25P1DataUnitID.getMessageLength()" because "this.mDataUnitID" is null

msullivan1993, 2025-08-15:
I'm not sure if this is related but I'm having multiple issues using multiple Airspy R2's as well.

Using 0.6.1 final to decode Ohio MARCS (because Nightly currently will not decode trunked P25 at all) one R2 is limited to 765-780MHz, other is limited to 850-860MHz (primarily to not switch to VHF) I currently have two towers in, Bartles and South Webster, which are both using 700MHz control channels. As long as radio traffic is on a 700MHz voice channel, there is no issue. However, the moment voice comes up on a 800MHz voice channel, both control channels lose lock and everything drops, then comes back. The same thing happens when I add in the 800MHz control channel for Greenup KY. All three control channels go idle. The moment I shut down Greenup, everything comes back on the other two.

I have also noticed this outside of standard traffic. If I view the spectrum on the 700MHz dongle, all is well, maintains lock, and decodes properly. The moment I bring up the 800MHz dongle spectrum, the control channels lose lock until I either disable spectrum, or switch back to the other dongle.

I have not attempted to recreate this issue using other dongles yet, that will be tomorrow's job.

jhorne-ferret, 2025-08-15:
No idea if this is the root of your issue, but the note about bringing up the spectrum on 700MHz and 800MHz devices simultaneously made me wonder if you're trying to drive both Airspy R2s at 10 MSPS from a single USB2.0 controller. To run both simultaneously at max sample rate, I suspect you'll need independent USB controllers.

msullivan1993, 2025-08-15:
> No idea if this is the root of your issue, but the note about bringing up the spectrum on 700MHz and 800MHz devices simultaneously made me wonder if you're trying to drive both Airspy R2s at 10 MSPS [...]

As far as I know that's not the issue, all of the USB ports and headers should be USB 3.1. The spectrum graph keeps rolling when it happens and I don't get any error messages. I can sit here and literally watch the 800MHz voice channel come up and Sync Loss messages start popping up until it drops.

I plan to test more tomorrow and see if it happens with other SDR's. I have two of everything basically (RSP1B, Nooelec SMArt, and RTL v4c) as well as try it on my main PC.

msullivan1993, 2025-08-19:
Quick update. Got a chance to test and play around. Using RTL SDR's I have no issues with decoding both 771MHz and 856MHz control channels at the same time, when the 856MHz control channel did float in. Same with my Nooelec SMArt's. But switching back to the Airspy, ran back into the same issue. So whatever the issue is, it's something regarding two Airspy SDR's simultaneously.

ikm19, 2026-02-22:
I just had similar Issues: I had 2 Airspy minis on one usb hub and one on the other, I tried to move them all to one hub, it stopped locking into control channels, today I separated one of the minis to a separate usb hub and it all came to normal. I think its a usb traffic limit or power issue, although I did add another power supply to a hub and that did not change the issue with 3 receivers on one hub.

DSheirer, 2026-11-07:
Closing.  sdrtrunk was updated with new jAirspy library that wraps the native airspy driver library.

closed 2026-11-07 by DSheirer