DSheirer/sdrtrunk #2243: MBE Recordings Should Include Additional Metadata

DSheirer, 2025-06-24:
**Is your feature request related to a problem? Please describe.**

As requested here: https://github.com/lwvmobile/dsd-fme/issues/298#issuecomment-2994414382

@DSheirer would if be possible to add an MFID string to the JSON formatted .mbe files in the future in SDRTrunk? Reason needed, is that on some encryption, its the only differentiation between how to decrypt (i.e., DMRA/Moto BP vs Hytera BP).

Edit: Also, would it be feasible to add system identifiers, such as NAC, SYSID, RFSS ID, WACN and so on for P25, if available, and CC and Trunking SYSCODE on DMR if available? Or too much work?

lwvmobile, 2025-06-24:
Also, just a followup to the MFID string or value, if placed in the .mbe file, very specifically for DMR, Ideally want the MFID of either:

1. Voice Link Control Frames prior to call, particularly if it is Hytera.
2. The MFID of the PI Header (if it is in the start of the call sequence).

If a PI header is present, it can supersede the MFID of the VLC frames, since it signals an enhanced privacy, and not basic.

Unfortunately, DMR Encryption is in such a state that you have to check down for at least 4 different possibilities, as opposed to just looking at one ESS field for P25. The possibilities include:

1. No PI Header, But SVC Options Signalling Encrypted Voice.
2. No Pi Header, But SVC Options Signalling Encrypted Voice and MFID 0x68 signalling Hytera.
3. PI Header Present, MFID 0x10 signalling DMRA Enhanced Privacy Modes (and SVC Options signalling enc).
4. PI Header Present, MFID 0x68 signalling Hytera Enhanced Privacy Modes  (and SVC Options signalling enc).

And that sadly, doesn't even cover the late entry mechanisms that both DMRA and Hytera use throughout a call.

Cagier, 2026-01-04:
I can get those error rates down as that was a quick test on a secondary antenna. I hadn't thought about the embedded IV so that is a really clever solution! (I would expect no less from you! 😉)

Are you using a modified version of DSD-FME with the -c option or did you create a standalone script for this? (I can probably produce something myself to do the same but it will take me a while to work out the details.)

Where the algo is predetermined and there is more than one (known) key then I guess trying the various keys and confirming which one produces the expected VC1 value would allow successful identification of the KeyID as well.

To get the required frame details, I have also previously used baseband NBFM recordings from SDRTrunk and fed those into DSD-FME with a batch file (which modifies the symbol rates to the required values and executes the program) but the results are quite mixed.  Using the mbe file would be much cleaner if it works so I will give that a try. Thanks again.

Obviously getting the full metadata from SDRTrunk for DMR at some point in the future will be much more reliable and efficient though...

Cagier, 2026-01-11:
To pursue this further I wrote code to do the following:

1) Process the AMBE Frames
2) Extract the relevant bits to get the C3 elements and produce some IV nybbles
3) Construct the IV from these individual Hex values
4) Loop through various starting points to verify which frame was actually the first frame in a Superframe
5) Calculate the LSFR values to help with that validation and also verify subsequent IVs
6) Create a revised mbe file with all the relevant parameters included so it could then hopefully be processed by DSD-FME

I pretty much got most of that done and produced output in the required format.  But then i realised that I had been using the wrong logic to get the C3 fragments - i.e. the logic that applies using the 1,4,7,2,5,8,3,6 sequence that applies to the raw 72 bit frames and not the finalised 49 bit AMBE values.  As I understand it, I should be actually constructing the 3 x 24 bit mi fragments from the 4 relevant (last?) bits of each 49 bit frame.  But I'm not sure if these are just simply concatanated or if there is further deinterlacing required?  Also then, I need to extract the relevant 12+12+8 IV bits/bytes based on a (24,12,8) golay decoding of those values which is more complex again.
I have been referring to the dsd_le.c source in DSD-FME but reversing or re-using that logic has been time-consuming so just wondering if there are is anything else available that could help with this or any other tips?  I think one short-cut might be to skip golay decoding and take the first 12 bits of each fragment but maybe I am deluding myself (and probably not as accurate).

Maybe this is not the right thread to be asking these questions so happy to take it elsewhere if appropriate...