DSheirer/sdrtrunk #2244: Crash

vsboost, 2025-06-24:
**sdrtrunk Version**
Nightly 2025-04-20

**Describe the bug**
Application crashed after running approx 1day and 18hrs

**Desktop (optional - complete the following information):**
 - OS: Ubuntu 25.04
 - CPU Cores: 4 cores]
 - RAM:16GB]

**Application Log**

Jun 24 15:48:47 sdr sdr-trunk.desktop[2179]: 2025-06-24 15:48:46.615 ERROR i.g.d.d.f.c.AbstractComplexPolyphaseChannelizer - Error removing polyphase channel source - channel so>
Jun 24 15:48:52 sdr sdr-trunk.desktop[2179]: java.lang.OutOfMemoryError: Java heap space
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: Uncaught exception:
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: java.lang.OutOfMemoryError
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: {0x000000070ff9f5e0} - klass: 'java/lang/OutOfMemoryError'
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - ---- fields (total size 5 words):
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private transient 'depth' 'I' @12  0 (0x00000000)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private transient 'backtrace' 'Ljava/lang/Object;' @16  null (0x00000000)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private 'detailMessage' 'Ljava/lang/String;' @20  "Java heap space: failed reallocation of scalar replaced objects"{0x000000070ff>
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private 'cause' 'Ljava/lang/Throwable;' @24  null (0x00000000)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private 'stackTrace' '[Ljava/lang/StackTraceElement;' @28  null (0x00000000)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]:  - private 'suppressedExceptions' 'Ljava/util/List;' @32  null (0x00000000)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # A fatal error has been detected by the Java Runtime Environment:
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #  Internal Error (upcallLinker.cpp:144), pid=2179, tid=2753
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #  Error: ShouldNotReachHere()
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # JRE version: OpenJDK Runtime Environment (23.0.2+9) (build 23.0.2+9)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # Java VM: OpenJDK 64-Bit Server VM (23.0.2+9, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # Problematic frame:
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # V  [libjvm.so+0xf9b863]  UpcallLinker::handle_uncaught_exception(oopDesc*)+0x43
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport -p%p -s%s -c%c -d%d -P%P ->
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # An error report file with more information is saved as:
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # /home/sdr/hs_err_pid2179.log
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: [68983.170s][warning][os] Loading hsdis library failed
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: # If you would like to submit a bug report, please visit:
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #   https://bell-sw.com/support
Jun 24 15:48:55 sdr sdr-trunk.desktop[2179]: #
Jun 24 15:48:55 sdr pipewire[1619]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-55) graph xrun not-triggered (1 suppressed)
Jun 24 15:48:55 sdr pipewire[1619]: pw.node: (alsa_output.pci-0000_00_1f.3.analog-stereo-55) xrun state:0x773748292008 pending:2/3 s:68994490245860 a:68994490336224 f:6899449034>
Jun 24 15:48:55 sdr pipewire[1619]: pw.node: (alsa_playback.java-63) xrun state:0x77374a710008 pending:0/1 s:68994500904069 a:68994490274943 f:68994490319458 waiting:18446744073>
Jun 24 15:48:55 sdr pipewire[1619]: pw.node: (alsa_playback.java-69) xrun state:0x773743ab5008 pending:0/1 s:68994500904069 a:68994490308118 f:68994490325750 waiting:18446744073>
Jun 24 15:48:56 sdr systemd[1567]: app-gnome-sdr\x2dtrunk-2179.scope: Consumed 1d 18h 58min 16.944s CPU time, 4.3G memory peak.
Jun 24 15:48:58 sdr sdrplay_apiService[1515]: [2752]: sdrplay_apiService_rsp: streamThread: Exit
Jun 24 15:48:58 sdr sdrplay_apiService[1515]: [2751]: sdrplay_apiService_device: mCmdHandlerThread: Exit
Jun 24 15:48:58 sdr sdrplay_apiService[1515]: [1528]: sdrplay_apiService: heartBeatThread: MSG: DevIdx0: RELEASED
Jun 24 15:49:11 sdr snapd[978]: storehelpers.go:916: cannot refresh: snap has no updates available: "bare", "core22", "desktop-security-center", "firefox", "firmware-updater", ">

**************************************************************

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  Internal Error (upcallLinker.cpp:144), pid=2179, tid=2753
#  Error: ShouldNotReachHere()
#
# JRE version: OpenJDK Runtime Environment (23.0.2+9) (build 23.0.2+9)
# Java VM: OpenJDK 64-Bit Server VM (23.0.2+9, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0xf9b863]  UpcallLinker::handle_uncaught_exception(oopDesc*)+0x43
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport -p%p -s%s -c%c -d%d -P%P -u%u -g%g -F%F -- %E" (or dumping to /home/sdr/core.2179)
#
[... 43 lines cut ...]
 handler table  [0x000079c4e82a2ba0,0x000079c4e82a2bd0] = 48
 scopes pcs     [0x000079c4e82a2bd0,0x000079c4e82a2c90] = 192
 scopes data    [0x000079c4e82a2c90,0x000079c4e82a2df0] = 352
RCX=0x0 is null
RDX=0x0000000000000001 is an unknown value
RSP=0x000079c4dcf00db0 is pointing into the stack for thread: 0x000079c4e4018690
RBP=0x000079c4dcf00dc0 is pointing into the stack for thread: 0x000079c4e4018690
RSI=0x0000000000000090 is an unknown value
RDI=0x000079c51edbb4c8: <offset 0x00000000011bb4c8> in /home/sdr/Apps/sdr-trunk-linux-x86_64-vnightly/lib/server/libjvm.so at 0x000079c51dc00000
R8 =0x000079c4e401aa80 points into unknown readable memory: 0x000079c4e003c928 | 28 c9 03 e0 c4 79 00 00
R9 =0x0000000000000001 is an unknown value
R10=0x0000000000000007 is an unknown value
R11=0x0 is null
R12=0x000000070ff9f5e0 is an oop: java.lang.OutOfMemoryError 
{0x000000070ff9f5e0} - klass: 'java/lang/OutOfMemoryError'
 - ---- fields (total size 5 words):
 - private transient 'depth' 'I' @12  0 (0x00000000)
 - private transient 'backtrace' 'Ljava/lang/Object;' @16  null (0x00000000)
 - private 'detailMessage' 'Ljava/lang/String;' @20  "Java heap space: failed reallocation of scalar replaced objects"{0x000000070ff9f608} (0xe1ff3ec1)
 - private 'cause' 'Ljava/lang/Throwable;' @24  null (0x00000000)
 - private 'stackTrace' '[Ljava/lang/StackTraceElement;' @28  null (0x00000000)
 - private 'suppressedExceptions' 'Ljava/util/List;' @32  null (0x00000000)
R13=0x000079c4e38c42e9 is pointing into metadata
R14=0x000079c4dcf00d68 is pointing into the stack for thread: 0x000079c4e4018690
R15=0x000079c4e4018690 is a thread

DSheirer, 2025-06-24:
The error indicates an out-of-memory situation and there are no visible cause(s) in the details that you posted.

Are you running any other software/services on this machine at the same time?  A quad-core 6th Gen i5 should be able to handle this workload, but if sdrtrunk is competing with other processes including the sdrplay API service, that could cause what you're seeing here.

If it looks like sdrtrunk is consuming a lot of available memory, try generating a thread dump so that we can see if there is a blocked thread that might be causing this.

vsboost, 2025-06-24:
Nothing else running besides rdio-scanner, i can attach the full dump that was generated if that helps, i only posted part of it in my original post as it was massive.

[dump.txt](https://github.com/user-attachments/files/20883537/dump.txt)

GitRDone2022, 2025-06-24:
I just had the same exact crash today on Windows.  I was able to repeat the crash after running for 10 minutes.  Found I had 1.4M log files in event_logs and java would hit 4GB of ram then crash.  After dumping most of the logs java memory use is ~ 600MB and program has been stable for 4 hours now.

I have the user option set to limit event logs to 500MB but it doesn't seem to dump the old logs.  I set-up a scheduled task to manually purge old logs as a workaround.

Maybe OP could check their event_logs folder size.

vsboost, 2025-06-26:
Since updating to the new nightly build, mem usage has been stable at around 1.7GB - SDRT has been running for approx 14hrs.  Will update the comments if anything crashes.

vsboost, 2025-06-27:
Another crash just now..

20250627 155128.584 [sdrtrunk channel [31/P25-1] 468200000 thread 1] ERROR i.g.d.d.f.c.o.ChannelOutputProcessor - Error processing channel results  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
20250627 155128.584 [sdrtrunk polyphase ifft processor thread 1] ERROR i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Error during IFFT and dispatch of processed channel results  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
20250627 155128.599 [sdrtrunk channel [47/P25-1] 421337500 thread 1] ERROR i.g.d.d.f.c.PolyphaseChannelSource - Error dispatching complex samples to listener [class io.github.dsheirer.sample.Broadcaster]  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
20250627 155128.595 [sdrtrunk channel [43/P25-1] 421637500 thread 1] ERROR i.g.d.d.f.c.PolyphaseChannelSource - Error dispatching complex samples to listener [class io.github.dsheirer.sample.Broadcaster]  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
20250627 155128.595 [sdrtrunk polyphase buffer processor thread 1] ERROR i.g.dsheirer.util.Dispatcher - Error while dispatching element [class io.github.dsheirer.buffer.airspy.AirspyNativeBuffer] to listener [class io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$NativeBufferReceiver]  [3GB/3GB 99%]
java.lang.OutOfMemoryError: Java heap space
	at ch.qos.logback.classic.Logger.buildLoggingEventAndAppend(Logger.java:424)
	at ch.qos.logback.classic.Logger.filterAndLog_0_Or3Plus(Logger.java:386)
	at ch.qos.logback.classic.Logger.error(Logger.java:543)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$NativeBufferReceiver.receive(PolyphaseChannelManager.java:532)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$NativeBufferReceiver.receive(PolyphaseChannelManager.java:479)
	at io.github.dsheirer.util.Dispatcher.process(Dispatcher.java:227)
	at io.github.dsheirer.util.Dispatcher$Processor.run(Dispatcher.java:250)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.runWith(Unknown Source)

adryd325, 2025-11-01:
Can reproduce

OS: Fedora 41 (Kernel 6.17.5)
CPU: Intel i7-6700TE (8 Cores) @ 3.40 GHz
RAM: 16 GB

[crash.txt](https://github.com/user-attachments/files/23283491/crash.txt)
[hs_err_pid4260.log](https://github.com/user-attachments/files/23283490/hs_err_pid4260.log)

Currently working on getting a heap dump

closed 2026-11-08 by DSheirer