DSheirer/sdrtrunk #2255: Parma, OH Simulcast won't decode

tylerwatt12, 2025-07-02:
**sdrtrunk Version**
sdrtrunk nightly 2025-06-30T07:26:04.739+0000

**Describe the bug**
Unable to decode system. Control channel is correct. I've tried LSM and C4FM, both result in SYNC LOSS.
Others have reported this same issue specifically with this software and exact radio system. I don't believe this has been reported as a bug yet.

One commenter mentioned he occasionally being able to receive, but only as C4FM, which doesn't make sense because Parma is a simulcast site. I suspect there is some quirk with this radio system causing inability to decode with sdrtrunk.

This radio system is shared between 3 different geographic areas. Parma, a city in Cuyahoga County (Phase 1), Medina County (25 miles south of Parma, P25 Phase 2), and Ottawa county (75 miles from Parma).

My APX 1000 has no problem listening to Parma, and Unitrunker has no trouble decoding the control channel and listing to audio.

I have not tried going to Medina and attempted to decode it with Sdrtrunk

https://forums.radioreference.com/threads/sdrtrunk-for-parma-ohio-use.457194/

**To Reproduce**
Attempt adding Parma/Medina/Ottawa TRS to SDRTrunk. Tried various methods, all result in SYNC LOSS

**Expected behavior**
Decoding of the control channel

**Screenshots**
[image]
[image]

**Application Log**

Baseband recording: 

[20250701_225010_853875000_Parma---Medina-County---Ottawa-County_Cuyahoga_Control_5_baseband.wav.zip](https://github.com/user-attachments/files/21010321/20250701_225010_853875000_Parma---Medina-County---Ottawa-County_Cuyahoga_Control_5_baseband.wav.zip)
**Desktop (optional - complete the following information):**
i7-8700
16GB RAM
Windows 10 Pro

**Additional context**
My signal for this site is as strong as any other sites i'm able to properly decode. I've ruled out low signal issues

tylerwatt12, 2025-07-06:
> I have been encountering a similar issue with the nightly builds. Do you by chance have one of the nightly builds from October of last year or earlier? When I use one of those versions everything works fine but when I use anything more recent all I get is sync-loss.

I don't have a nightly from then to try. Only what's available on this git.

tylerwatt12, 2025-07-10:
I've played with the last 4 releases and here are my notes:

1: For this Parma system, the control channel changes very frequently, several times a day
2: [Parma CC Decoding does not work in the nightly or 0.6.1 final release, works on beta3 and lower. All further notes pertain to beta3 or older only](https://youtu.be/NJN6JvsWuaY?t=1231)
3: [Decoding only works with C4FM, even though RR shows it LSM/Simulcast](https://youtu.be/NJN6JvsWuaY?t=660)
4: [Decoding consistently works when adding the current control channel frequency in the top spot](https://youtu.be/NJN6JvsWuaY?t=628)
5: Frequency rotation delay of 400 is too short, setting it to 2000ms sometimes helps it latch onto the CC better (when the current CC is in the top position)
6: [And even still, sometimes the results are inconsistent](https://youtu.be/NJN6JvsWuaY?t=945)

tylerwatt12, 2025-07-29:
This is a Phase 1 control channel system. (Both Parma and Medina sites). Matching the control channel frequencies to the bandplan I use on the APX radio confirm this

tadscottsmith, 2025-07-30:
I’d recommend removing all tuners except one and focusing on locking in the control channel first. It looks like some of your channels are assigned a preferred tuner while others are not.

Skyking-Bravo, 2025-08-08:
Has there been any progress made on this issue? I am to the point where I am going to have to purchase Airspy devices so I can use Trunk-Recorder. I am currently using all SDRplays at the monitoring site where I am having the problems and unfortunately Trunk-Recorder doesn't work well with SDRplay. I have tried everything imaginable, but I get the same result everytime... versions after 10/21/24 only produce sync loss.

Skyking-Bravo, 2025-08-08:
> I've played with the last 4 releases and here are my notes:
>
> 1: For this Parma system, the control channel changes very frequently, several times a day 2: [Parma CC Decoding does not work in the [...]

My experience matches your notes.

msullivan1993, 2025-08-15:
Going to bump this up. I'm having issues with P25 trunked decoding on the latest Nightly as well (Aug 2025) I have not been able to decode the closest MARCS tower (Bartles/Coal Grove, about 4 miles from me)

If I downgrade to 0.6.1 beta 3 or 0.6.1 final, I have no problem decoding the Bartles control channel on any device (tested with RSP1B, Airspy R2, and RTL v4c) and everything works as it should (including adding traffic), but if I try on Nightly, it locks on for about 2-3 seconds then fades and goes into sync loss. Occasionally it will come back then drop right out again. 

Nightly has also had an issue with P25 conventional on a singular channel here, but I'm not 100% sure if it's an issue with Nightly or if it was an antenna issue I was having. Will update further once I test overnight.