DSheirer/sdrtrunk #2262: LibUsb - Error [-4 LIBUSB_ERROR_NO_DEVICE]

Tonywpa, 2025-07-24:
**sdrtrunk Version**
SDRTrunk V 0.6.0/ Also getting the same error with 0.6.1.

**Describe the bug**
sdrtrunk does not see RTL-SDR v4.

**To Reproduce**
Steps to reproduce the behavior:
1. install drivers using Zadig (bulk-in, Interface (Interface 0) with WinUSUB driver v6.1.7600.16385
2. Install JDK22 (Unzipped in same folder as sdrtrunk.
3. right click sdr-trunk.bat->run as administrator
4. Nothing on waterfall. Logs shows  "ERROR i.g.d.s.t.m.TunerManager - LibUsb - error [-4/LIBUSB_ERROR_NO_DEVICE] during get device list  [9MB/512MB 1%]"

**Expected behavior**
sdrtruck would see dongle.

**Screenshots**

[image]

**Application Log**

[image]

**Desktop (optional - complete the following information):**
 - OS: Windows11
  - RAM:16GB

**Additional context**
The RTL-SDR works with 2 other programs- Airspy and SkyRoof. 
I am brand spanking new to this, be gentle.

DSheirer, 2025-07-24:
1. the sdrtrunk download already includes Java -- please remove the JDK 22 install that you added - it's not needed.
2. try moving the tuner dongle to other USB ports and see if sdrtrunk can see it then.
3. report back here what you find?

Tonywpa, 2025-07-24:
Ok.  I deleted everything and started from the beginning, without the JDK 22.  I successfully got it running on my laptop that is running Win10.  I'm having the issue on my PC running Win11.
I looked in the task manager for multiple instances of sdrtrunk, only 1 showing. 
I tried 5 different usb ports.
AirSpy and Skyroof still work.  I'm at a loss.

DSheirer, 2025-07-25:
I don't know what's causing this and looking across the internet, I'm not finding many leads to go on.

Tonywpa, 2025-07-25:
Understood.  I saw this issue was brought up in this forum a couple of times, but the OS was Linux.  I'm not giving up, when I find the fix, I'll reply back.  It will be a while.  Thank you for your efforts.

DSheirer, 2026-11-07:
Closing as resolved in new jRTL library.

closed 2026-11-07 by DSheirer