DSheirer/sdrtrunk #2278: NBFM noise/squelch settings enhancement

srsanford, 2025-08-26:
**Is your feature request related to a problem? Please describe.**
I record and stream a mix of NBFM and P25 channels. Since the August 3, 2025 NBFM noise/squelch changes, noisy transmissions from unintended sources are being recorded and streamed. Prior to these squelch enhancements, I could increase the squelch to a value above the noisy signals. The Aug 3 enhancements do work to eliminate/reduce the squelch tails which is great... 

**Describe the solution you'd like**
The squelch panel allows Noise to be controlled between 6.2 and 8. Enhance the squelch controls so that the Noise range can go higher than 8.0 (i.e. closer to 10). I believe this would allow me to set Noise Open to a value of 9.6 or so and eliminate the noise.

**Describe alternatives you've considered**

1. I considered that I may not fully understand the new settings. Maybe there is a way to control the squelch to my needs, however, I have not figured out to date. Any insights are appreciated.
2. Implement a CTCSS decoder. I acknowledge this has been requested before.

**Additional context**
Baseband recording intending to record East Hartford Fire Operations on 460.600
[20250823_162643_460600000_East-Hartford_East-Hartford_East-Hartford-Fire-Operations_58_baseband.wav](https://drive.google.com/file/d/1AXYVwDhDug61Nv1-DZrD4vVqTW5jCN9Z/view?usp=sharing)

Screenshots

Frequency query for 460.600 in CT

[image]

Example screenshot during the noise (460.600). Same problem occurs with 2 other channels I stream. It also occurs with all my CMED channels, however, I do not stream them. I also source the 3 problem channels via a BCT15X scanner (with PL's defined) and do not pick up the noisy signals.

[image]

msullivan1993, 2025-08-27:
I have noticed some random open squelching on some transmissions as well but because I use this to monitor railroad traffic, which is all analog CSQ simplex communication, it's been hard to decipher between the software false detecting and a very distant transmission. All settings left default per Dennis's recommendation.

It's infinitely better than "the old system" but it could use a bit more tweaking. 

(And maybe some optimization.. that graph is murder on CPU/GPU usage. 😄 )

Randesign, 2025-11-26:
No need to bump. I started to research ways of implementing CTCSS, but just haven't had the time to implement. It's on my todo list (pretty far down at the moment - months away). Maybe another developer will get to it before I do. SDRtrunk is a large codebase and relatively few contributors for such a huge list of feature requests. You'll just have to be patient, or learn Java and contribute, or point another developer here.

closed 2026-09-26 by DSheirer