DSheirer/sdrtrunk #2285: Disabling second Airspy R2 crashes SDRTrunk

msullivan1993, 2025-09-29:
**sdrtrunk Version**
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.
Nightly and Final

**Describe the bug**
A clear and concise description of what the bug is.
When using two Airspy R2's, one Airspy has no issues enabling/disabling tuner, but attempting to disable the second R2 (IE for troubleshooting an issue) crashes SDRTrunk. 

**To Reproduce**
Steps to reproduce the behavior:
Load SDRTrunk with two R2's. Disable one, then disable the other. One of the two will cause the crash.

**Expected behavior**
A clear and concise description of what you expected to happen.
Tuner disabled...

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: Windows 10
 - CPU Cores: 8 core 16 thread (Ryzen 7 3700X)
 - RAM: 32GB DDR4-3200

**Additional context**
Add any other context about the problem here.
Nothing really else to add.

msullivan1993, 2025-09-29:
[hs_err_pid5772.log](https://github.com/user-attachments/files/22589437/hs_err_pid5772.log)

Referenced error file

DSheirer, 2026-11-05:
With the switchover to jAirspy library, this is hopefully resolved.  Please advise if this issue is still there and we can reopen this issue.

closed 2026-11-05 by DSheirer